mbridak/not1mm #232: reature req- new window Summary

JimW3FA, 2024-12-01:
**Describe the solution you'd like**
Is it possible to add a "Summary" window as below that can be opened with a shortcut (DXLog uses Alt S), user enabled/disabled.

**Additional context**
Shortcut: [Alt][S]

[image]

The summary window displays the number of QSO (which does not include the dupes), along with multipliers relevant
to the contest type for each band in the blue section, with the dark blue strip showing the current band being operated.

Important: The Points and Multiplier counts may not include bonuses whereas the total score does.

Important: The QSO count includes X-QSO but excludes dupes.

Grey section at the bottom shows totals for the contest and final raw score.

mbridak, 2024-12-02:
Looks useful and do-able.
I may need clarification on what a X-QSO is.

py4xx, 2025-02-07:
> Preciso de esclarecimentos sobre o que é um X-QSO.

Hi Mike... 73...
X-QSO, in DXLog, is when you log a contact, but you don't want it in the log anymore, or due to an error in the contact, or any other reason. But the DXLog log does not delete the contact. Then you use X-QSO, (If I'm not mistaken it's CTRL + X) and then X-QSO appears in the log which makes the cabrillo understand that that contact is null.
I'm kind of bad at explaining things... If you still have questions, call me.

mbridak, 2025-04-03:
I'm thinking this is close-able with the statistics window.

closed 2025-04-03 by mbridak