mbridak/not1mm #233: feature req- Rate window

JimW3FA, 2024-12-01:
Here is the link to DXLog documentation which is easier than trying to describe.
Little background, DXLog.net is my preferred logging software on Windows, but since switching my machine to Manjaro Linux and finding "not1mm" my preferred is now to live in a Linux world using not1mm.

Rate

Shortcut: [Alt][R]

[image]

The panel displays live QSO rate data for the current contest log.

The top section shows QSO's in last hour, last 10 QSO rate, last 100 QSO rate,
rate for the current hour so far and, if enabled in the context menu, a projected score.

There is also a menu option for additionally showing the rate over the past 10 minutes.

The second section shows time on and time off in hours and minutes. The minimum off time
defined in the rules for the current contest is used in this calculation.

Important: Since DXLog does not know the starting time or duration of a contest,
the on/off time calculation starts and ends with the first and last QSO. This means that
if you do not join the contest from start, or end early, you will need to add this off time
time to the displayed off time manually.

The third section shows Run and S&P statistics. Both in total QSO count and the count since
the last full hour.

For a contest entry with band change limitations, the next section displays these.

For a contest with QRB-based scoring, the next section displays distance statistics.

The second to lowest section shows time since last multiplier, multiplier and QSO points worth.

Important: The calculation of multiplier points does not include factors like bonuses.

The last section shows operating mode and audio focus status.

The right-click menu allows you to configure the Rate window.

Rate rc.png

Target QSO per hour is the expected number of QSO per hour. When set, the Rate window will display the
different between the resulting target QSO count and the actual QSO count for the current hour.
It resets at the top of the hour.

The global QSO rate is the expected number of QSO per hour during the entire contest. When set,
the Rate window will display the difference in QSO count and rate versus this target.

Rate targets.png

You can set the displayed rate unit.

Rate ru.png

It is also possible to display the rate per band or mode.

Ratepermodeband.png

mbridak, 2024-12-02:
I'll put it on the todo list.

mbridak, 2025-01-23:
@JimW3FA , I've merged in a rate window.

[image]

It's a start. I'll try and add more if I can come to understand what some of the definitions are.