mbridak/not1mm #257: Feature Request - Change Mark color in Dark Mode

FastThenLeft, 2025-01-26:
The dark blue mark color is almost invisible in Dark Mode on the dark background of the band map. Please change it to something that you can see!

Thank you for such an AWESOME program! You have made Linux contesting fun again!

-David, N9KT

mbridak, 2025-01-26:
Sure, not a problem. There's a small catch. I'm colorblind, so things that look fine to me don't look fine to other people. So I'll end up choosing something equally hard for you to see.

So your mission, if you choose to accept it is to give me RGB values for the color you want it to be. It must also be easily visible in non-dark mode.

mbridak, 2025-01-26:
`rgb(245, 194, 17)` maybe this

FastThenLeft, 2025-01-27:
Michael-
Your suggestion of RGB(225,127,0) looks great!
Thanks again!
-David, N9KT

mbridak, 2025-01-27:
No Problem. It's pushed out.

closed 2025-01-27 by mbridak