mbridak/not1mm #269: Crash on start after update 24.12.29 to 25.1.29

raufaser, 2025-02-06:
**Describe the bug**
Starting the application in a shell. Getting the following stacktrace:

```
Traceback (most recent call last):
  File "/home/me/.local/bin/not1mm", line 8, in <module>
    sys.exit(run())
             ^^^^^
  File "/home/me/.local/lib/python3.11/site-packages/not1mm/__main__.py", line 4026, in run
    window = MainWindow(splash)
             ^^^^^^^^^^^^^^^^^^
  File "/home/me/.local/lib/python3.11/site-packages/not1mm/__main__.py", line 650, in __init__
    self.rate_window = RateWindow()
                       ^^^^^^^^^^^^
  File "/home/me/.local/lib/python3.11/site-packages/not1mm/ratewindow.py", line 48, in __init__
    uic.loadUi(fsutils.APP_DATA_PATH / "ratewindow.ui", self)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/load_ui.py", line 86, in loadUi
    return DynamicUILoader(package).loadUi(uifile, baseinstance)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/PyQt6/uic/Loader/loader.py", line 62, in loadUi
    return self.parse(filename)
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 1014, in parse
    self._handle_widget(ui_file.widget)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 842, in _handle_widget
    self.traverseWidgetTree(el)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 818, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 280, in createWidget
    self.traverseWidgetTree(elem)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 818, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 280, in createWidget
    self.traverseWidgetTree(elem)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 818, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 512, in createLayout
    self.traverseWidgetTree(elem)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 818, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 555, in handleItem
    self.traverseWidgetTree(elem)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 818, in traverseWidgetTree
    handler(self, child)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 271, in createWidget
    self.stack.push(self._setupObject(widget_class, parent, elem))
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/PyQt6/uic/uiparser.py", line 237, in _setupObject
    self.wprops.setProperties(obj, branch)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/properties.py", line 428, in setProperties
    getattr(self, prop_name)(widget, prop)
  File "/usr/lib/python3/dist-packages/PyQt6/uic/properties.py", line 513, in orientation
    {'Qt::Horizontal': QtWidgets.QFrame.Shape.HLine,
KeyError: 'Qt::Orientation::Horizontal'
```

Before the update all worked fine. I just logged a QSO and wanted to update after that. Then i encountered the bug on the next start.

**To Reproduce**
Steps to reproduce the behavior:
1. Start applcation 25.1.29

**Expected behavior**
The application starts

**Screenshots**
see stacktrace

**Desktop (please complete the following information):**
 - OS: Debian GNU/Linux
 - Version 12 (bookworm)

mbridak, 2025-02-06:
Heavy Sigh...

Seems debian uses an older, to no surprise, version of uiparser...
I'll edit the tags out by hand and re-push.

mbridak, 2025-02-06:
@raufaser Okay I pushed a new version lemme know if it fixed it.

raufaser, 2025-02-06:
Works very well. Fixed.
Thank you :)

closed 2025-02-06 by raufaser