mbridak/not1mm #284: not1mm not starting

JimW3FA, 2025-03-24:
**Describe the bug**

[w3fa@contest ~]$ not1mm
Traceback (most recent call last):
  File "/usr/bin/not1mm", line 5, in <module>
    from not1mm.__main__ import run
  File "/usr/lib/python3.13/site-packages/not1mm/__main__.py", line 28, in <module>
    import notctyparser
ModuleNotFoundError: No module named 'notctyparser'
[w3fa@contest ~]$ 

**To Reproduce**
Steps to reproduce the behavior:
either open application from desktop icon or menu icon

**Expected behavior**
not1mm to open normally without error

**Screenshots**
Text from terminal screen is above

**Desktop (please complete the following information):**
 - OS: Manjaro x64
 - Version latest (rolling updates)

**Additional context**
This problem occurred after a previous not1mm update. This is the latest update that I did today (24March2025).

[w3fa@contest ~]$ pamac build not1mm-git
Preparing...
Cloning not1mm-git build files...
fatal: not a git repository (or any of the parent directories): .git
hint: Using 'master' as the name for the initial branch. This default branch name
hint: is subject to change. To configure the initial branch name to use in all
hint: of your new repositories, which will suppress this warning, call:
hint:
hint: 	git config --global init.defaultBranch <name>
hint:
hint: Names commonly chosen instead of 'master' are 'main', 'trunk' and
hint: 'development'. The just-created branch can be renamed via this command:
hint:
hint: 	git branch -m <name>
hint: Using 'master' as the name for the initial branch. This default branch name
hint: is subject to change. To configure the initial branch name to use in all
hint: of your new repositories, which will suppress this warning, call:
hint:
hint: 	git config --global init.defaultBranch <name>
hint:
hint: Names commonly chosen instead of 'master' are 'main', 'trunk' and
hint: 'development'. The just-created branch can be renamed via this command:
hint:
hint: 	git branch -m <name>
Generating not1mm-git information...
Checking not1mm-git dependencies...
Warning: downgrading package not1mm-git (r1815.572d45c-1 => r1447.5cad407-1)
Resolving dependencies...
Checking inter-conflicts...

To build (1):
  not1mm-git  r1447.5cad407-1  (r1815.572d45c-1)  AUR

Edit build files : [e] 
Apply transaction ? [e/y/N] y

==== AUTHENTICATING FOR org.manjaro.pamac.commit ====
Authentication is required to install, update, or remove packages
Authenticating as: W3FA (w3fa)
Password: 
==== AUTHENTICATION COMPLETE ====

Building not1mm-git...
==> Making package: not1mm-git r1447.5cad407-1 (Mon 24 Mar 2025 01:42:20 PM EDT)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> Retrieving sources...
  -> Cloning not1mm-git git repo...
Cloning into bare repository '/var/tmp/pamac-build-w3fa/not1mm

mbridak, 2025-03-24:
I myself have not been able to install from AUR with pamac for awhile now using my manjaro kde from a vm.

Not sure if it will help or if it's what you're looking for but in a pinch I use uv to run the latest version of not1mm on various platforms.

You can install uv with:
`curl -LsSf https://astral.sh/uv/install.sh | sh`

then run not1mm with:
`uv tool run not1mm`

mbridak, 2025-03-25:
@JimW3FA , okay I reinstalled my manjaro vm.

I did find a problem with the project license tag in the pyproject toml file which prevented it from building for some reason.
After correcting, it installed with pamac and ran fine on my end.

JimW3FA, 2025-03-25:
Ok thanks for the update. I initially thought it was my machine as no one
else seemed to make mention of any issues. I will try it today and let you
know. Hoping to use it in CQ WPX this weekend. Ended up using yfktest last
contest which I'm not to crazy about.

Jim

closed 2025-04-03 by mbridak