mbridak/not1mm #300: ESM mode combining F2/F3 doesn't work for me

awandahl, 2025-04-10:
**Describe the bug**
Ehhh... A new computer and a new installation of not1mm. Of some reason the combination of F2/F3 in RUN ESM mode doesn't send the F2 part = macro "R|F2|HisCall|{HISCALL}". Contest is CQWW CW and I think I have tried everything. The F2 key separately does send {HISCALL} fine, but not when combining the F2/F3.

**Desktop (please complete the following information):**
 - OS: xubuntu 24.04
 - Version v25.4.10

**Additional context**
Add any other context about the problem here.

mbridak, 2025-04-11:
Not sure what's going on. It works fine here...

These are the macros I used:

 ```text
R|F2|HisCall|{HISCALL}
R|F3|Run Exch|{SNT} {EXCH}
```

awandahl, 2025-04-11:
Exactly what I have, but now I made a work around by setting this:
```
R|F2|HisCall|{HISCALL}
R|F3|Run Exch|{HISCALL} {SNT} {EXCH}`
```

mbridak, 2025-04-12:
Fixed Issue with small timing delay.

awandahl, 2025-04-14:
Works fine now, thanks!

closed 2025-04-14 by mbridak