mbridak/not1mm #301: Auto CQ indication

awandahl, 2025-04-10:
Very good that Auto CQ is an option! Thanks!

It's very good to have an indication that auto CQ is enabled. The green dot is useful but I would prefer a countdown would be better. It's kind of a standard I think. So if you have a three seconds delay between CQ:s the countdown would start at 3.0 and then count down to 2.5  2.0  1.5  1.0  0.5 and send the next CQ. This will also be a reminder of what delay you have configured.

Additionally the delay in seconds I have set in the config seems to have very little to do with real seconds. I don't know why, but I have very hard to find the setting I want (3 seconds).

And yet another observation is that at times haven't been able to stop the Auto CQ by any means. It may have to do with that I'm using a Winkeyer clone along with K6GTE winkeyerserial. Maybe there is some buffering. ESC doesn't stop the Auto CQ, typing in the callsign field neither, and closing not1mm doesn't help. Turning off the radio seems to work though. A mystery...

Using xubuntu 24.04 and v25.4.10.1

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

awandahl, 2025-04-10:
If I put 11 secs delay in the configuration I get about 3 seconds delay in real life

mbridak, 2025-04-10:
There's no pollable way to ask cwdaemon etc to ask if it's done sending. So
the delay is a blind timer to refire the F1 macro.
So your macro must take about 8 seconds to key, leaving to 3 seconds pause.

mbridak, 2025-04-10:
I believe if you just tap either side of the paddle connected to your winkeyer, it will stop sending anything in it's buffer.

awandahl, 2025-04-11:
OK, I think I understand now. In a real world operating situation I will run CQ (automatically or manually), it's not much to do, condx are bad and I'm half asleep. I may have heard something interesting, and at the same time I have pressed the CQ button or the automatic CQ has already started. The panic button is ESC but the transmission will not end until the CQ has ended not matter how hard I hit the keyboard, right? The solution for now is to hit the paddle (that works fine). Is cwdaemon and rigctld also buffering, do you know that? ESC as panic button would be convenient as most operating is very keyboard centered.

mbridak, 2025-04-12:
@awandahl I've updated winkeyerserial to add the clear buffer command I saw today.
I've updated it to PyQt6.
Pushed version to pypi, so you can update with pip or pipx or uv... however you installed it.

Updated Not1MM, and pushed to pypi as well.

I tested it and seems to work as expected.

mbridak, 2025-04-12:
@awandahl Okay, Since I had my winkeyer out to test the rewrite I decided to try your problem with the F2 macro not sending.

I turned on all the debugging messages in the CW class. 
The both macros were being sent to the winkeyerserial program.
And since they both sent fine when manually pressing F2 then F3, the only difference I could think of was the speed at which the ESM routine sent the macros to the keyer. So I added a 2/10ths a second delay between sending the macros.
and the problem went away...

So pull v25.4.11.3 and give it a whirl.
Hope it works for you. 

Cheers Mike K6GTE

mbridak, 2025-04-12:
@awandahl Okay, I added the visual indicator.

[image]

mbridak, 2025-04-14:
Ha, Well yeah... I'll fix it.

mbridak, 2025-04-14:
Okay give 25-4-14-1 a try.

mbridak, 2025-04-19:
I'm calling this one complete.

closed 2025-04-19 by mbridak