mbridak/not1mm #302: Add Scandinavian Activity Contest SAC

awandahl, 2025-04-10:
Plenty of time, it's in September. I ran it with N1MM 2024. I prefer NOT next time.

mbridak, 2025-04-11:
```text
Scandinavian Activity Contest, CW
Status:	Active
Geographic Focus:	Scandinavia
Participation:	Worldwide
Mode:	        CW
Bands:	        80, 40, 20, 15, 10m
Classes:	        Single Op (QRP/Low/High)
                        Scandinavian: Single Op Single Band
                        Non-Scandinavian: Single Op Low Band
                        Single Op Overlays: (TB-Wires/Rookie/Wires Only/Classic)
                        Multi-Single
                        Multi-Multi
                        Explorer Single-Op
                        Explorer Multi-Op
Max power:	HP: >100 watts
                        LP: 100 watts
                        QRP: 5 watts
Exchange:	RST + Serial No.
Work stations:	Once per band
QSO Points:	Scandinavian: 2 points per QSO with Europe
                        Scandinavian: 3 points per QSO with non-Europe
                        Europe: 1 point per QSO with Scandinavian station
                        non-Europe: 1 point per QSO with Scandinavian station on 20, 15, 10m
                        non-Europe: 3 points per QSO with Scandinavian station on 80, 40m
Multipliers:	Scandinavian: Each DXCC country once per band
                        non-Scandinavian: Each Scandinavian call area once per band
Score Calculation:	Total score = total QSO points x total mults
Upload log at:	http://www.sactest.net/robot/
Mail logs to:	(none)
Find rules at:	https://www.sactest.net/blog/scandinavian-activity-contest-2024-rules/
Cabrillo name:	SAC-CW
```

awandahl, 2025-04-11:
This is probably the most difficult part:  

"8.2 For non-Scandinavian stations

Each worked prefix-number (Ø-9) in each Scandinavian DXCC entity is valid for one multiplier on each band (e.g. SI3, SK3, SL3, SM3, 7S3, and 8S3 all count for the same (1) multiplier). Stations without a prefix number count for the 0th area (e.g., LA/G3XYZ counts for LAØ), and those with two or more prefix numbers counts for the area indicated by the first number (e.g., OZ150A counts for OZ1)."

mbridak, 2025-04-11:
That does seem challenging.

mbridak, 2025-04-11:
Would you happen to have a Cabrillo log from a past event? It would help having a list of valid calls from the area to test with.

awandahl, 2025-04-11:
My log would not be very illustrative anyhow since I get my multipliers by DXCC, operating from Scandinavia.

mbridak, 2025-04-11:
Okay loaded someones cabrillo with about 480 Q's so I could test with and I think this is a winner for the Scandinavian mult part.

```sql
SELECT count(DISTINCT(CountryPrefix || ":" || substr(WPXPrefix,3,1))) as mults from DXLOG where ContestName = 'SAC-CW' AND CountryPrefix IN ('JW', 'JX', 'LA', 'LB', 'LC', 'LG', 'LI', 'LJ' , 'LN', 'OF', 'OG', 'OH', 'OI', 'OFØ', 'OGØ', 'OHØ', 'OJØ', 'OX', 'XP', 'OW', 'OY', '5P', '5Q', 'OU', 'OV', 'OZ', '7S', '8S', 'SA', 'SB', 'SC', 'SD', 'SE', 'SF', 'SG', 'SH', 'SI', 'SJ', 'SK', 'SL', 'SM', 'TF');
```

The dxcc per band one is a no brainer.

mbridak, 2025-04-12:
Okay, I added SAC CW and SSB.
You have a long time to test it.

awandahl, 2025-04-14:
Will do! Both as Scandinavian and DX station. The final counting is made by a robot at the organizer but it's nice if the multipliers shown in not1mm is reasonable. Thanks a lot!

mbridak, 2025-04-19:
Closing this as complete.

closed 2025-04-19 by mbridak