mbridak/not1mm #312: Contest Sent Exchange field is unclear (how to setup contest exchange)

j0uni, 2025-04-21:
I'm trying to setup a contest with exchange that should consist of "RST 001 PM" in my case, serial number and province.  I guess only contest that supports this is UKEIDX?

I cannot get F3 to send "5NN TT1 PM" no matter what I try to put in.

I think this would need more explanation in the documentation. There is very little (or none) documentation on this field, what kind of patterns it supports.

j0uni, 2025-04-21:
If I enter "# PM" in the contest exchange field, the UI shows "001 PM" in sentNR - field but send "5NN # PM" as CW when I press F3. The error is that it sends # and not the actual 001 (or TT1).

mbridak, 2025-04-21:
Your answer can be found by [Looking Here](https://github.com/mbridak/not1mm?tab=readme-ov-file#ukei-dx).

j0uni, 2025-04-22:
Thanks for the answer! A bit hacky way it seems.

I think that it sends # in CW is a bug (it should send the number)?

Any change NRAU - style competition could be added? https://www.nraubaltic.eu/rules

closed 2025-04-23 by mbridak