mbridak/not1mm #318: winkeyerserial and symlinks?

awandahl, 2025-04-30:
**Describe the bug**
Can winkeyerserial use symlinked ports? I made an entry in udev where I hardcoded my Winkeyer K3NG clone to /dev/ttyWinkeyer, but it doesn't seem to work. When I remove the SYMLINK+="ttyWinkeyer" in the udev and run straight on /dev/ttyUSB0 it works fine. I symlink all my radios in udev to keep a bit of order, SYMLINK+="ttyQMX", SYMLINK+="ttyK2", SYMLINK+="ttyUbitx" etc. Would be nice to do the same with the winkeyer and the not1mm vfo box... In the future it would also be nice if it would be possible to choose rig from within not1mm. I run one process of rigctld with separate port numbers in the background for each radio.

**To Reproduce**
Use udev to hardcode the winkeyer to a designated port and try to use winkeyerserial with not1mm.

**Desktop (please complete the following information):**
 - OS: xubuntu 24.04
 - Version 25.4.28

**Additional context**
Not a not1mm problem I guess, but it affects the setup. And a nice surprise when fiddling with this was that I realized that I could key my K2 via CAT from not1mm. Didn't know before that worked. I have replaced the RS232 interface with USB C in my K2. Maybe that did the trick. Waiting for CAT keying for my QMX too...

mbridak, 2025-05-01:
Okay I've updated winkeyerserial. I myself don't have any symlinked devices. So you'll have to tell me if it worked.
I had to update the program anyways to remove the use of a deprecated library.

awandahl, 2025-05-01:
Works now, thanks!

closed 2025-05-02 by mbridak