mbridak/not1mm #338: Rotor support....

awandahl, 2025-06-11:
**Describe the bug**
When starting up 25.6.11 I get
````
QImage::pixelColor: coordinate (0,0) out of range
QImage::setPixelColor: color is invalid
QImage::pixelColor: coordinate (0,0) out of range
QImage::setPixelColor: color is invalid
QImage::pixelColor: coordinate (0,0) out of range
QImage::setPixelColor: color is invalid
QImage::pixelColor: coordinate (0,0) out of range
QImage::setPixelColor: color is invalid
Setting heading
````
and sparse compass decoration.

**Desktop (please complete the following information):**
 - OS: xubuntu 24.04
 - Version 25.6.11

**Additional context**
Add any other context about the problem here.

mbridak, 2025-06-11:
Got around to running this in a VM with your distro. 

Got the same stream of QImage::pixelColor: coordinate (0,0) out of range

No map at all. So what I think is happening is that for some reason it's not loading the rectangular map. So the QImage object has no size and there is no pixel at 0,0 to read from.

Okay I mysteriously seemed to fix it. Don't know how.

awandahl, 2025-06-12:
Yiiihaaa! Indeed, a mysterious solution or not, it worked. I'm very happy with the sparse and stylish map. No need for colors. Thanks!

mbridak, 2025-06-12:
I was curious and tried the look ups on HamQTH, sm0hpl returned:

[image]

but 5h3wa returned:

[image]

Got back no valid grid (So no valid heading) and no name.

So I at-least have to account for invalid returns from HamQTH.