mbridak/not1mm #356: Problem with Cabrillo export of IARU HF contest

mlforcada, 2025-07-13:
Some entries of my IARU HF contest log are wrongly exported to Cabrillo by not1mm version 25.7.12. The ADIF does not contain these errors.

How can I reproduce the error? I'd be happy to provide you with the database.

I have the suspicion that the error affects entries for which I corrected the sent exchange in the UI after finding a mistake, just after completing the QSO. Looks like the Cabrillo is generated from the uncorrected entries.

 I run not1mm on Debian GNU/Linux 12.

Thanks!

mlforcada, 2025-07-13:
I used a Xiegu G90, with the CAT cable that came with it, only to control frequency (I hear and send CW manually).
Can't seem to be able to attach the .db file here ("file type not allowed").  I will email it to you.

mbridak, 2025-07-13:
Okay, Just pushed 25.7.13.1 to PyPi. This should fix your issue. The Cabrillo generator was pulling the zone from the Zone field in the DB and not the Nr field which is what you logged. Thanks again for reporting this, I would never have caught this otherwise.

mlforcada, 2025-07-14:
Thanks a million!! That was fast!

closed 2025-07-15 by mbridak