mbridak/not1mm #375: "Enter" shortcut disregards S&P / Running info.

aknrdureegaesr, 2025-07-28:
## Describe the bug

I have different info in my CW templates for S&amp;P and running. I use the "Enter" shortcut to send CW messages. This shortcut disregards whether I'm in  S&amp;P or Running mode.

## To Reproduce

1. Set up CW messages that are different for S&amp;P and running.
2. Set to running, type a call sign, use "Enter" to send the messages and log the QSO.
3. Set to S&amp;P and repeat.

## Expected behavior

The CW messages that get send are different and reflect the state, whether `not1mm` is in S&amp;P or running mode.

## Desktop

 - OS: Debian GNU Linux bookworm
 -  `not1mm` version f5c03e0da3b7932f74e11d43b535c16678e7e36c

## Context

The "Enter" shortcut is extremely helpful, particularly in contests such as the IARU HF championship when the response can be predicted by the program.

mbridak, 2025-07-29:
Which contest did you have loaded at the time?  Since the ESM logic is stored in each contest plugin.

aknrdureegaesr, 2025-07-29:
I was trying out `not1mm` in the context of the IARU HF championship. (All my issues I entered July 28 were in that contest's context.)

mbridak, 2025-07-29:
I can't replicate what you're describing.
Works fine for me.